HinTak/seeed-voicecard #14: Missing taskbar issue

phantomrider24, 2023-10-14:
Hi there,
I recently bought a audio hat 4 mic linear array board. I've installed the git repository from the actual seeed site but they refuse to install due to them being outdated. I found this reposity to help my previous issue but when i install the drivers from this reposity, my taskbar seems to disappear. I have to have the taskbar present for my intended use of the raspberry pi. Please can someone help me out

NarrowSpace, 2023-10-21:
I had the same issue when I was using the most recent OS version Bookworm, despite that, everything else works properly. However, once I downgraded to Bullseye, the driver could not be detected, only the LED light was working.... What OS version are you using btw.

NarrowSpace, 2023-10-21:
I had the same issue when I was using the most recent OS version Bookworm, despite that, everything else worked properly. However, once I downgraded to Bullseye, the driver could not be detected, only the LED light was working... What OS version are you using btw.

NarrowSpace, 2023-10-21:
I had exactly the same issue before, once I rebooted the taskbar started flashing and just gone forever.

I reinstalled the system a couple of times. Then I realized that if I didn't change the position of the taskbar and just left it at the top, it wouldn't disappear after I rebooted the Pi after I installed this git repo. I'm not sure if this solution only works for me or if it's a small bug in this version.

BTW, I'm using the most recent system version, OS Bookworms 64-bit.

NarrowSpace, 2023-10-21:
I had exactly the same issue before; once I rebooted, the taskbar started flashing and then gone forever.

I reinstalled the system a couple of times. Then I realized that if I left the taskbar at the top, it wouldn't disappear after I rebooted the Pi, after installing this Git repo. I'm not sure if this solution works only for me or if it's a small bug in this version.

BTW, I'm using the most recent system version: OS Bookworms 64-bit.

phantomrider24, 2023-10-24:
hi,
I am using debain bullseye legacy 64-bit os

HinTak, 2024-01-10:
Yes, probably a good idea keeping this open, until the info is captured somewhere noticeable.

- This was documented elsewhere: you can stop lxpanel from crashing if you uninstall alsa-pulseaudio, I think. Search here or upstream on closed issues about 'lxpanel'.

NarrowSpace, 2024-01-17:
I recently reinstalled the latest Bookworm version (64-bit, 2023.12.05), and the taskbar was missing again. I solved the issue by directly deleting PulseAudio, rebooted, and the taskbar came back.

HinTak, 2024-02-05:
It is mentioned in https://github.com/respeaker/seeed-voicecard/issues/319#issuecomment-1430929683

HinTak, 2024-02-05:
The actual procedure is in https://raspberrypi.stackexchange.com/questions/122579/after-fresh-install-of-raspberry-os-the-menu-bar-is-missing-in-tightvnc-session