HinTak/seeed-voicecard #18: Cant get 4-mic hat to record

Technorocker, 2023-12-05:
Hello @HinTak I appreacite you taking the time to continue development on seed devices. I have been scouring the web for the past week trying to get my 4-mic hat array working on the bullseye rasp OS, Ive tried both 32 and 64 bit versions, I finally decided to stick with the 32 bit hoping for better compatibility but to now avail. I am using your repo now and although the system does recognize the drivers I cant for the life of me find a proper command to get it to record any audio. 

I even tried to pump data into chatgpt since it is aware of respeaker and its products it claims but that gpt is becoming dumber and dumber by the day it seems. 
Here are a few details to my setup:
arecord -L gives me

```
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault
    Default Audio Device
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
speex
    Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Playback/recording through the PulseAudio sound server
speaker
dmixer
array
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
hw:CARD=ArrayUAC10,DEV=0
    ReSpeaker 4 Mic Array (UAC1.0), USB Audio
    Direct hardware device without any conversions
plughw:CARD=ArrayUAC10,DEV=0
    ReSpeaker 4 Mic Array (UAC1.0), USB Audio
    Hardware device with all software conversions
sysdefault:CARD=ArrayUAC10
    ReSpeaker 4 Mic Array (UAC1.0), USB Audio
    Default Audio Device
front:CARD=ArrayUAC10,DEV=0
    ReSpeaker 4 Mic Array (UAC1.0), USB Audio
    Front output / input
dsnoop:CARD=ArrayUAC10,DEV=0
    ReSpeaker 4 Mic Array (UAC1.0), USB Audio
    Direct sample snooping device
usbstream:CARD=ArrayUAC10
    ReSpeaker 4 Mic Array (UAC1.0)
    USB Stream Output
```
I noticed when I try to run a simple arecord command i get a channel 

chatgpt suggested i create a .asoundrc file with this for its contents

```
pcm.!default {
    type asym
    playback.pcm "speaker"
    capture.pcm "dmixer"
}

[... 120 lines cut ...]
and at the bottom this stuff

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'r' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2097
Expression 'PaAlsaStreamComponent_FinishConfigure( &self->capture, hwParamsCapture, inParams, self->primeBuffers, realSr, inputLatency )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2730
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2842
Traceback (most recent call last):
  File "/home/rhasspy/./micCheck1.py", line 44, in <module>
    record_audio(respeaker_index, "test.wav")
  File "/home/rhasspy/./micCheck1.py", line 9, in record_audio
    stream = p.open(
             ^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pyaudio/__init__.py", line 639, in open
    stream = PyAudio.Stream(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pyaudio/__init__.py", line 441, in __init__
    self._stream = pa.open(**arguments)
                   ^^^^^^^^^^^^^^^^^^^^
OSError: [Errno -9999] Unanticipated host error

Have you seen or come across any of this or am I cursed? lol :(

HinTak, 2023-12-05:
Wrong driver - why do you have a USB audio driver installed? The 4-mic device in this repo is connected via the 40-pin hat. There are two 4-mic devices supported by this driver - a linear shape one and a square looking one, both mounted on top on the pi via the 40-pin hat. There is a picture of at least one of them if not both in the front.

HinTak, 2023-12-05:
There is a picture in the front with both of them. If your device looks like one of those, you have installed the wrong software, not the code here, and you should follow the install procedure here. If your device does not look like one of those, then you have simply come to the wrong place, and you should go where you got your device and your software from, and seek help there.

closed 2023-12-05 by HinTak

Technorocker, 2023-12-05:
Hey hintak, thanks for your quick reply. I assumed the board 4-mic hat was the same as their consumer sold 4-mic hat just in a case
i am using the 4-mic hat array they put into a single device https://wiki.seeedstudio.com/ReSpeaker-USB-Mic-Array/

HinTak, 2023-12-05:
Told you already - different device, different driver. If you don't believe my word, what else can I say? Half (really, more than half) of the driver code is about connectivity. USB connection is definitely not 40-pin hat.

HinTak, 2023-12-05:
And it is not even the same device wrapped in a different case/connection. Did you even look at the picture at the front and compare?

Technorocker, 2023-12-05:
holy dude, relax,
respeaker sells a 4-mic hat array dev board and a 4-mic hat array in a case using usb. I was led to the respeaker github https://github.com/respeaker/seeed-voicecard which led me to your repo as an updated version to use since respeaker quit supporting their product. My mistake and sorry to question your superiority over all things respeaker. 

my mistake for not being a linux coding guru like yourself  and trying to learn new things. I bow down to your intellect. Sorry to have wasted your time.

HinTak, 2023-12-05:
I'll repeat your own url for you https://wiki.seeedstudio.com/ReSpeaker-USB-Mic-Array/  - thar uses linux's generic USB audio driver, not the code here. Hardware-wise , that has a XVF-3000. The 4-mics devices here uses either one ac108 or a pair of ac108 plus a ac101. Please go away. You are in the wrong place. And, I don't work for seeedstudio.