HinTak/seeed-voicecard #22: Kernel Panic trying to test with arecord

belzedaar, 2023-12-28:
I am using the Seeed Respeaker 4 mic hat:  !https://files.seeedstudio.com/wiki/ReSpeaker-4-Mic-Array-for-Raspberry-Pi/img/connect2.jpg!

The SD card died in my Pi3b voice assistant so I thought I'd try and upgrade it to a Pi4 I was using elsewhere.

Flash latest official Pi Lite version to SD card:
```uname -a
Linux vision 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux```

git clone https://github.com/HinTak/seeed-voicecard.git

The default branch is v6.1 so good to go there.

sudo ./install.sh and reboot

After reboot:
arecord -L 
```hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, bcm2835-i2s-ac10x-codec0 ac10x-codec0-0
    Direct hardware device without any conversions```

Try and record as per the instructions on the wiki:
```arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav```

Kernel Panic when pressing Ctrl+C to stop recording
![20231228_150718](https://github.com/HinTak/seeed-voicecard/assets/6566875/5d5e42c4-a795-49ae-b4d7-f2cfdf977ce8)

I tried using the linked (super old) version of Buster with the drivers pre-installed (https://files.seeedstudio.com/linux/Raspberry%20Pi%204%20reSpeaker/2021-05-07-raspios-buster-armhf-lite-respeaker.img.xz)

That works correctly and I can record, sox to stereo and playback via headphones as expected.

I tried using 32 bit instead of 64 bit, bit that doesn't seem to install correctly, 
```cannot get card index for seeed4micvoicec``` when trying to record.
I also tried 64 bit full desktop RPi image in case somehow that was using a different kernel, but no luck there.

It seems lots of people are working OK with 6.1 kernels from various forms around the place, so I am really not sure what is going wrong in my case.

Any help would be greatly appreciated (via donation!)

Weaverzhu, 2024-01-23:
same here

removing `-Dac108` seems to solve the problem for `arecord`, but a lot of other functions will still cause the kernel panic (e.g. pyaudio)

nahuelproietto, 2024-04-02:
same issue... 
I'm using kernel 5.15.0-1049-raspi with ubuntu-22.04 server arm64 (raspberry pi 4 model B)