HinTak/seeed-voicecard #30: Issue: Support for latest raspios bookworm

nahuelproietto, 2024-04-02:
Hey everyone, I'm unable to install the **4mic-array** drivers using latest raspios (raspios_lite_arm64-bookworm).
[https://downloads.raspberrypi.com/raspios_lite_arm64/images/.../2024-03-15-raspios-bookworm-arm64-lite.img.xz](https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-03-15/2024-03-15-raspios-bookworm-arm64-lite.img.xz).  

#### Overview

I'm getting the following message when running the install script from the **6.6 branch**. Seems like dkms and i2c-tools has not installation candidates on raspios-bookworm. I also tried installing: https://github.com/respeaker/seeed-voicecard
Please let me know if anyone is facing something similar,

```
uname -r
6.6.20+rpt-rpi-v8
```

```
git clone -b v6.6 https://github.com/HinTak/seeed-voicecard
cd seeed-voicecard && sudo ./install.sh

...

Package i2c-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package dkms is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'dkms' has no installation candidate
E: Package 'i2c-tools' has no installation candidate
./install.sh: line 151: dkms: command not found
./install.sh: line 153: dkms: command not found
setup git config
git init
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint: 
hint: 	git config --global init.defaultBranch <name>
hint: 
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint: 
hint: 	git branch -m <name>
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) d3535f3] origin configures
 7 files changed, 1482 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!

```

After rebooting, I'm getting an empty list of capture hardware devices
```
$ arecord -l
$ **** List of CAPTURE Hardware Devices ****
```

giagar5, 2024-04-02:
exact same issue here, no solution yet

devinhenkel, 2024-04-29:
Same. Any tips to get it working? I love my 4-mic array and would love to get it working.

devinhenkel, 2024-04-30:
I can get through the install with Ubuntu, but I get a "Failed to load module i2s-mmap" error on boot. Such a brilliant piece of hardware, I want to make it work...

devinhenkel, 2024-05-13:
So, I installed OVOS and it's working. Need to dig in and figure out what they're doing differently.

HinTak, 2024-05-23:
Pi 5 is a different issue: https://github.com/HinTak/seeed-voicecard/issues/19 ; kernel 6.6/6.8 uses the v6.6/v6.8 branches.

closed 2024-05-23 by HinTak