HinTak/seeed-voicecard #31: 2-Mic Hat doesn't work on Raspberry Pi 4 Model B

hongfeij, 2024-04-06:
Hi:

I encountered such issue when running command from official document to test the recording function:

```
hongfeij@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 3: seeed2micvoicec [seeed-2mic-voicecard], device 0: fe203000.i2s-wm8960-hifi wm8960-hifi-0 [fe203000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

hongfeij@raspberrypi:~ $ arecord -D "plughw:3,0" -f S16_LE -r 16000 -d 5 -t wav test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
**arecord: set_params:1416: Unable to install hw params:**
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 4000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 16000
TICK_TIME: 0

```

Any clue about the reason and solution?

willumpie82, 2024-04-08:
I run into exactly the same error, 
tried different forks (started with the one from the getting started at seeed-studio wiki, but that one is really old

planned to create a small kiosk with wyoming satallite, waveshare display 4.3" DSI display, respeak 2mic and custom case

beitong95, 2024-04-09:
sudo rpi-update 1440b3e0b52075a9ec244216cddcf56099c28dfa to downgrade your kernel to 6.1.21 works for me. also I use the v6.6 branch. I don't know why

HinTak, 2024-05-23:
Yes, use the right branch.

closed 2024-05-23 by HinTak

dihan, 2025-01-15:
Took my along time to find this and even then had some issues getting it to work. Anyone else comes across this issue i've created a basic guide as well as one line install for a Pi 4 (porbably others too) with 2 Mic Hat on the following repo

https://github.com/dihan/pi-voice-assistant

HinTak, 2025-01-15:
Please use upstream for any issues/postings.