HinTak/seeed-voicecard #36: [FR] Refactor to Debian packaging

fonix232, 2024-05-23:
Hi @HinTak - great work on keeping things up to date!

There's one thing that bothers me a little. The current structure, while sticks closely to the original Seeed repo, is not really suitable for long term support, or easy installation, especially with all the separate branches for different kernel versions.

I'd like to recommend a refactor that would introduce a proper build system to create Debian packages. This would allow a number of convenience features:

- Easier distribution - APT repos can easily hosted on GitHub, and users wouldn't need to run install/uninstall scripts via sudo. This would also allow for easier updates.
- More fine-grained installations. By packaging components separately (e.g. a `a10x-dev` package containing the `a10x.h` file, separate DKMS driver packages for `a101`, `a108` and `wm8960`, as well as a separate `seed-voicecard` DKMS package, culminating in semi-meta packages for the various card models), one could install just what is needed to get their device working.
- Support for multiple kernel versions in the same branch. No more dozen+ branches for each kernel versions, but rather each module would have multiple subfolders, targeting different kernel versions, keeping everything in one place yet still separate (e.g. `a108/src/6.2/a108.[hc]` etc.)
- Automated packaging using GHA Workflows. Just update the files, merge into `main`, tag the version, and bam, new packages published.
- Remove the need of manual compilation and manual handling of overlays, etc., by having the Debian package take care of it
- The new folder architecture could then be used for other distros' package builders as well, making it possible to deploy packaging for Arch etc.
- Remove the need to maintain the hacky install script, by breaking it down into separate parts for the components, making maintenance much simpler

Ideally this would result in the following packages:

- seeed-a10x-dev - containing only the `a10x.h` file to be used for both the build process and development
- seeed-a101-dkms - the DKMS drivers for the A101
- seeed-a108-dkms - the DKMS drivers for the A108
- seeed-wm8960-dkms - the DKMS driver for the WM8960
- seeed-voicecard-2mic - package containing the files necessary for the 2-mic voicecard (DTBO, ALSA config and state), with a DependsOn set as 
- seeed-voicecard-4mic - package for the 4-mic voice card
- seeed-voicecard-6mic/8mic - same as above

And with a bit of tinkering, the DKMS install script could be modified to even match the appropriate kernel X.Y folder within and compile that code (although I think version-specific changes should be done within the source, e.g. how it's done at https://github.com/HinTak/seeed-voicecard/blob/v6.1/ac108.c#L1328-L1332)

fonix232, 2024-05-23:
FYI I did start such a refactor, but all the scattered branches make it hard to pick a starting point, so your input would be most welcome regarding that :)

HinTak, 2024-05-23:
There are historically ways of suuporting multiple kernels via dkms : https://github.com/HinTak/seeed-voicecard/tree/v6.1/patches - I think it still works if you suddenly want to reboot into a much older 4.x kernels. It was just getting uglier and uglier when upstream kernel source changes. You basically need a diff for each of the distinct branches, and keep the diff in one places, and test each of them, by rebooting into a dozen different older kernels from time to time when you add a new one. It was an approach that proved to be painful.

Please feel free to spend your time on it, but, please do not try to decide how I spend my time.

closed 2024-05-23 by HinTak

fonix232, 2024-05-24:
Please note that nowhere in my original post did I specify that I'd want _you_ to make these changes. I simply brought it up as a possible improvement, not to demand you to spend your time on it, but to gauge interest and possibly co-operate on implementing it, without placing any immediate pressure on you to actually provide this feature. Apologies if it came through as if I was demanding it.

As for multi-kernel support with the same DKMS package - Debian packaging does support changing how the DKMS install script runs, so by creating a module structure such as this:

```
seeed-voicecard/
├── ac108/
│   ├── 6.6/
│   │   ├── ac108.c
│   │   └── ac108.h
│   ├── 6.1/
│   │   ├── ac108.c
│   │   └── ac108.h
│   ├── 5.5/
│   │   ├── ac108.c
│   │   └── ac108.h
│   └── 4.9/
│       ├── ac108.c
│       └── ac108.h
└── wm8960/
    ├── 6.5/
    │   ├── wm8960.c
    │   └── wm8960.h
    ├── 6.3/
    │   ├── wm8960.c
    │   └── wm8960.h
    ├── 5.9/
    │   ├── wm8960.c
    │   └── wm8960.h
    ├── 5.1/
    │   ├── wm8960.c
    │   └── wm8960.h
    └── 4.7/
        ├── wm8960.c
        └── wm8960.h
```

it should be possible to both package, and compile the right version for the kernel. The only downside I see is that a kernel update would not trigger a proper rebuild (as the buildscript would need to reside in the `postinst` of the DKMS package), though I think it is possible to subscribe to package update events on Debian, which in turn could trigger a manual run of the `postinst` script. 

Also, please don't misunderstand the above as me forcing any kind of work on you - I'm purely brainstorming here, in public, so that further input from other interested parties can be gathered. For this purpose, the only thing I'm asking is that you leave this "issue" open, so that others may find it, even if you have no interest in undertaking any part of it.

HinTak, 2024-05-24:
That's not how I understand dkms works. That said, the multiple-directory layout, one per kernel version, is very easy to do: you just have multiple git submodules pointing to, instead of different repos, just different branches of the same repo (this). Please feel free to maintain such a meta package elsewhere.